cornell-zhang/bnn-fpga #1: Could you show the CPATH you included for compilation?

PAPL-SKKU, 2017-03-18:
I'm trying to build your codes with SDSoC 2016.1 (also 2016.3)
but, I failed to build it when I followed the instructions in README.
I think it's because I set wrong C_INCLUDE_PATH before make. 
Could you list the C_INCLUDE_PATH (or CPATH) you've used?

Thank you

seanlatias, 2017-03-18:
Hi,

To make SDSoC work, you only need to source "settings64.sh", which should be under the folder of SDSoC/2016.1 (or 2016.3). The CPATH is used if you want to specify Vivado or Vivado_HLS version. For  example,

export CPATH=PATH_TO_VIVADO/Vivado/2016.4/include:$CPATH

If possible, would you please show the error messages? Thanks!

garion9013, 2017-03-18:
Oops, I've logged in with wrong ID. I'm the same person above.
On my machine -- Ubuntu 14.04, gcc 4.8.4, SDSoC 2016.1 --
I sourced settings.64.sh into env.
And my PATH/CPATH settings.
```
export PATH=/opt/Xilinx/SDSoC/2016.1/SDK/2016.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:$PATH
export PATH=/opt/Xilinx/SDSoC/2016.1/bin:$PATH
export PATH=/opt/Xilinx/SDSoC/2016.1/SDK/2016.1/bin:$PATH
export PATH=/opt/Xilinx/SDSoC/2016.1/Vivado_HLS/2016.1/bin:$PATH

export CPATH=/opt/Xilinx/SDSoC/2016.1/Vivado/2016.1/include/:$CPATH
```

I typed following commands
```
cd bnn-fpga/cpp/accel
make
```
so ```accel_test_layer.exe, accel_test_random.exe, accel_test_bnn.exe``` are succesfully popped out.

but, to generate bitstream
```
cd bnn-fpga/cpp/accel/sdsoc_build
make
```

gives following error messages
```
...

INFO: [SDSoC 0-0] Compiling /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/accel_test_bnn.cpp
                 from /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/accel_test_bnn.cpp:9:
../../utils/../minizip/zip.h:23:20: fatal error: zlib.h: No such file or directory
 #  include "zlib.h"
                    ^
compilation terminated.
ERROR: [SDSoC 0-0] Exiting sds++ : Error when calling 'arm-linux-gnueabihf-g++ -c -I../../utils -DHLS_COMPILE -O3 -std=gnu++11 -g -D __SDSCC__      -I /opt/Xilinx/SDSoC/2016.1/aarch32-linux/include  -I /opt/Xilinx/SDSoC/2016.1/Vivado_HLS/2016.1/include /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/accel_test_bnn.cpp -o accel_test_bnn.o'
sds++ log file saved as /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/sdsoc_build/_sds/reports/sds_accel_test_bnn.log

make: *** [accel_test_bnn.o] Error 1
```

So, I manually add ```/usr/include/``` to CPATH
then it passes but different error is occurred

```
/usr/include/bits/mathinline.h: In function ‘int round_away_from_zero(float)’:
/usr/include/bits/mathinline.h:776:19: error: invalid 'asm': invalid operand for code 'w'
       : "0" (__x));           \
                   ^
/usr/include/bits/mathinline.h:500:5: note: in definition of macro ‘__inline_mathcodeNP_’
     code;              \
     ^
/usr/include/bits/mathinline.h:761:1: note: in expansion of macro ‘__inline_mathcodeNP’
 __inline_mathcodeNP (ceil, __x, \
 ^
/usr/include/bits/mathinline.h:758:19: error: invalid 'asm': invalid operand for code 'w'
       : "0" (__x));           \
                   ^
/usr/include/bits/mathinline.h:500:5: note: in definition of macro ‘__inline_mathcodeNP_’
     code;              \
     ^
/usr/include/bits/mathinline.h:743:1: note: in expansion of macro ‘__inline_mathcodeNP’
 __inline_mathcodeNP (floor, __x, \
 ^
ERROR: [SDSoC 0-0] Exiting sds++ : Error when calling 'arm-linux-gnueabihf-g++ -c -I../../utils -DHLS_COMPILE -O3 -std=gnu++11 -g -D __SDSCC__      -I /opt/Xilinx/SDSoC/2016.1/aarch32-linux/include  -I /opt/Xilinx/SDSoC/2016.1/Vivado_HLS/2016.1/include /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/AccelTest.cpp -o AccelTest.o'
sds++ log file saved as /home/arc-yhlinux/sdx_workspace/bnn-fpga/cpp/accel/sdsoc_build/_sds/reports/sds_AccelTest.log

make: *** [AccelTest.o] Error 1
```

Any ideas?
Thanks

rzhao01, 2017-03-19:
Googled around, the issue is that sds++ is trying to cross-compile to ARM but by adding /usr/include to CPATH you are forcing the native glibc header files. Are you adding /usr/include at the beginning or end of CPATH? Adding it at the end should work.

garion9013, 2017-03-20:
Nope. Putting it last doesn't solve the problem, same errors.
It seems that I include wrong ```zlib.h``` while arm cross-compiling
I cannot find ```zlib.h``` in ```/opt/Xilinx/SDSoC/```
Could I ask where your ```zlib.h``` is in your machine?

garion9013, 2017-03-20:
OK. I downloaded [zlib 1.2.11](http://www.zlib.net/) source codes,
and set CPATH to include the directory. And solved the issues!
The error may be because I wrongly include zlib header files while cross-compile.

Thank you guys for all the responses and your extraordinary works about BNN.
Good luck!

rzhao01, 2017-03-20:
Hmm, I guess by putting zlib in another directory you can access the zlib files without including /usr/include. Thanks for your fixes to the code as well!

closed 2017-03-20 by rzhao01