cornell-zhang/bnn-fpga #3: zlib.h is not found

SnailWalkerYC, 2017-07-06:
When I run the project with accel_test_bnn.cpp, I meet with the following error: how to deal with it? I run the project on Win7 Platform and use HLS 2017.1.
.../bnn-fpga-master/bnn-fpga-master/cpp/accel/../minizip/zip.h:23:20: fatal error: zlib.h: No such file or directory compilation terminated.

rzhao01, 2017-07-06:
The code is looking for the zlib library, which is a fairly standard library for working with zip archives. I'm not sure how to install zlib on Windows 7; I personally have only ran the code on CentOS Linux.

SnailWalkerYC, 2017-07-06:
Thank you. I found the source file on Github: https://github.com/madler/zlib and fixed this problem on Windows 7. 

But the compiler of 2017.1 HLS of Win 7 still has bugs, which is not compatible with c++0x and c++11 or above. I asked Xilinx PAE Engineer, they told it was the software bug and didn't fix until 2017.3 HLS. I need to change the gramma like "**static constexpr const char* filename = "/data/cifar10_test_inputs.zip";**" to c++98 version. like char filename[50] = "/data/cifar10_test_inputs.zip"; or char * filename = "/data/cifar10_test_inputs.zip";  Maybe I should change the platform to Linux.

SnailWalkerYC, 2017-07-06:
I changed little about the code and it can work with C Simulation now.

rzhao01, 2017-07-07:
I don't think the code works without C++11 support. I don't remember the exact reason but there was some feature which made it necessary.

If you can use Linux instead I would heavily recommend it. As I said I've only ever tried the code on Linux.

closed 2018-03-07 by rzhao01