cornell-zhang/bnn-fpga #4: can't compile because of gcc version?

MinLiAmoy, 2017-07-25:
hi, my SDSoC version is 2016.4 on Ubuntu 16.04, and i just follow the instruction in README. At first, the log shows something wrong with zlib.h. I downloaded the source and it seems to be resolved. But i still cannot compile the code.
the log message is:

> make -C minizip
> make[1]: Entering directory '/home/ml/bnn-fpga-master/cpp/minizip'
> cc -c -O -I../.. -DHAVE_AES miniunz.c
> cc -c -O -I../.. -DHAVE_AES unzip.c
> cc -c -O -I../.. -DHAVE_AES ioapi.c
> cc -c -O -I../.. -DHAVE_AES minizip.c
> unzip.c: In function ‘unz64local_GetCurrentFileInfoInternal’:
> unzip.c:898:17: warning: passing argument 3 of ‘unz64local_getByte’ from incompatible pointer type [enabled by default]
>                  if (unz64local_getByte(&s->z_filefunc, s->filestream_with_CD, &uL) != UNZ_OK)
>                  ^
> unzip.c:198:11: note: expected ‘int *’ but argument is of type ‘uLong *’
>  local int unz64local_getByte(const zlib_filefunc64_32_def* pzlib_filefunc_def, voidpf filestream, int *pi)
>            ^
> unzip.c:902:17: warning: passing argument 3 of ‘unz64local_getByte’ from incompatible pointer type [enabled by default]
>                  if (unz64local_getByte(&s->z_filefunc, s->filestream_with_CD, &uL) != UNZ_OK)
>                  ^
> unzip.c:198:11: note: expected ‘int *’ but argument is of type ‘uLong *’
>  local int unz64local_getByte(const zlib_filefunc64_32_def* pzlib_filefunc_def, voidpf filestream, int *pi)
>            ^
> unzip.c:907:17: warning: passing argument 3 of ‘unz64local_getByte’ from incompatible pointer type [enabled by default]
>                  if (unz64local_getByte(&s->z_filefunc, s->filestream_with_CD, &uL) != UNZ_OK)
>                  ^
> unzip.c:198:11: note: expected ‘int *’ but argument is of type ‘uLong *’
>  local int unz64local_getByte(const zlib_filefunc64_32_def* pzlib_filefunc_def, voidpf filestream, int *pi)
>            ^
> cc -c -O -I../.. -DHAVE_AES zip.c
> cc -O -I../.. -DHAVE_AES -o miniunz miniunz.o unzip.o ioapi.o libaes.a -lz
> cc -O -I../.. -DHAVE_AES -o minizip minizip.o zip.o   ioapi.o libaes.a -lz
> ar rv ./libminizip.a miniunz.o unzip.o minizip.o zip.o ioapi.o
> r - miniunz.o
> r - unzip.o
> r - minizip.o
> r - zip.o
> r - ioapi.o
> ar rv ./libminizip.a miniunz.o unzip.o minizip.o zip.o ioapi.o
> r - miniunz.o
> r - unzip.o
> r - minizip.o
> r - zip.o
> r - ioapi.o
> ranlib ./libminizip.a
> make[1]: Leaving directory '/home/ml/bnn-fpga-master/cpp/minizip'
> make -C utils
> make[1]: Entering directory '/home/ml/bnn-fpga-master/cpp/utils'
> g++ -c DataIO.cpp -o DataIO.o -O3 -std=gnu++11 -g
> g++ -c ParamIO.cpp -o ParamIO.o -O3 -std=gnu++11 -g
> g++ -c ZipIO.cpp -o ZipIO.o -O3 -std=gnu++11 -g
> g++ -c Timer.cpp -o Timer.o -O3 -std=gnu++11 -g
> cc1plus: error: unrecognized command line option ‘-std=gnu++11’cc1plus: error: unrecognized command line option ‘-std=gnu++11’
> 
> cc1plus: error: unrecognized command line option ‘-std=gnu++11’
> cc1plus: error: unrecognized command line option ‘-std=gnu++11’
> Makefile:14: recipe for target 'DataIO.o' failed
> make[1]: *** [DataIO.o] Error 1
> make[1]: *** Waiting for unfinished jobs....
> Makefile:14: recipe for target 'ZipIO.o' failed
> make[1]: *** [ZipIO.o] Error 1
> Makefile:14: recipe for target 'ParamIO.o' failed
> make[1]: *** [ParamIO.o] Error 1
> Makefile:14: recipe for target 'Timer.o' failed
> make[1]: *** [Timer.o] Error 1
> make[1]: Leaving directory '/home/ml/bnn-fpga-master/cpp/utils'
> Makefile:3: recipe for target 'all' failed
> make: *** [all] Error 2

is it something wrong with the version of c++? #3 
thanks!

seanlatias, 2017-07-26:
Which Makefile are you running? I assume you're running the Makefile under accel because there is no SDSoC command involved. In this case, it is not related to #3. What you should do is to upgrade your C compiler version. You can also try with "-std=c++11".  If the new flag does not work, can you tell me which g++ version are you using? Thanks!

MinLiAmoy, 2017-07-26:
I ran the Makefile under cpp folder. I have updated my g++ version to 4.8. But after source 'settings64.sh', the compiler version turns out to be 4.6 which is under SDSoC environment.

> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/opt/Xilinx/SDx/2016.4/Vivado_HLS/lnx64/tools/gcc/bin/../libexec/gcc/x86_64-unknown-linux-gnu/4.6.3/lto-wrapper
> Target: x86_64-unknown-linux-gnu
> Configured with: ../gcc-4.6.3/configure --prefix=/proj/autoesl/tools/gcc/4.6.3_x86_64 --enable-languages=c,c++ --disable-checking --enable-__cxa_atexit --disable-libunwind-exceptions
> Thread model: posix
> gcc version 4.6.3 (GCC) 
If i didn't source settings64.sh, it still has some errors.

> make -C minizip
> make[1]: Entering directory '/home/ml/bnn-fpga-master/cpp/minizip'
> make[1]: Nothing to be done for 'all'.
> make[1]: Leaving directory '/home/ml/bnn-fpga-master/cpp/minizip'
> make -C utils
> make[1]: Entering directory '/home/ml/bnn-fpga-master/cpp/utils'
> g++ -c DataIO.cpp -o DataIO.o -O3 -std=gnu++11 -g
> g++ -c ParamIO.cpp -o ParamIO.o -O3 -std=gnu++11 -g
> g++ -c ZipIO.cpp -o ZipIO.o -O3 -std=gnu++11 -g
> g++ -c Timer.cpp -o Timer.o -O3 -std=gnu++11 -g
> In file included from DataIO.h:8:0,
>                  from DataIO.cpp:1:
> Common.h:7:20: fatal error: ap_int.h: No such file or directory
>  #include <ap_int.h>
>                     ^
> compilation terminated.
> g++ -c Common.cpp -o Common.o -O3 -std=gnu++11 -g
> In file included from ParamIO.cpp:5:0:
> Common.h:7:20: fatal error: ap_int.h: No such file or directory
>  #include <ap_int.h>
>                     ^
> compilation terminated.
> Makefile:14: recipe for target 'DataIO.o' failed
> make[1]: *** [DataIO.o] Error 1
> make[1]: *** Waiting for unfinished jobs....
> In file included from Common.cpp:1:0:
> Common.h:7:20: fatal error: ap_int.h: No such file or directory
>  #include <ap_int.h>
>                     ^
> compilation terminated.
> Makefile:14: recipe for target 'ParamIO.o' failed
> make[1]: *** [ParamIO.o] Error 1
> Makefile:14: recipe for target 'Common.o' failed
> make[1]: *** [Common.o] Error 1
> make[1]: Leaving directory '/home/ml/bnn-fpga-master/cpp/utils'
> Makefile:3: recipe for target 'all' failed
> make: *** [all] Error 2
> 
Thanks!

rzhao01, 2017-07-26:
That looks like it can't find the Vivado HLS headers. Can you check you'be
setup thE Vivado paths correctly?

MinLiAmoy, 2017-07-26:
Thanks! I downgrade SDSoC to 2016.4 but forgot to change the CPath correctly. Now it works. Thanks a lot for your help.

closed 2017-07-27 by zhangzhiru