cornell-zhang/bnn-fpga #5: max_align_t problem

qingzengsong, 2017-08-16:
when mak in cpp，I have a problem ‘’max_align_t‘’.
In file included from /opt/Xilinx/SDx/2016.4/Vivado_HLS/include/gmp.h:53:0,
                 from /opt/Xilinx/SDx/2016.4/Vivado_HLS/include/floating_point_v7_0_bitacc_cmodel.h:143,
                 from /opt/Xilinx/SDx/2016.4/Vivado_HLS/include/hls_half.h:40,
                 from /opt/Xilinx/SDx/2016.4/Vivado_HLS/include/ap_int.h:28,
                 from ../utils/Typedefs.h:4,
                 from AccelTest.h:4,
                 from AccelTest.cpp:1:
/usr/include/c++/5/cstddef:51:11: error: ‘::max_align_t’ has not been declared
   using ::max_align_t;
           ^

I add #include <cstddef> to 
ParamIO.cpp
Common.cpp
DataIO.cpp
AccelTest.cpp
  can solve this problem.

But, this wether lead other problem, since 
'ParamIO.cpp
Common.cpp
DataIO.cpp
AccelTest.cpp'
will be build by Vivado_HLS?

MinLiAmoy, 2017-08-17:
I met the same problem. I added #include <cstddef> and the problem was solved. There's no problem with the SDSoC.

qingzengsong, 2017-08-17:
> I met the same problem. I added #include and the problem was solved. There's no problem with the SDSoC.

But I have the problem since #include with the sdsoc.
There is error，

发自我的 iPhone

qingzengsong, 2017-08-17:
Pragma processor failed: In file included from /home/songqingzeng/bnn-fpga-master/cpp/accel/Accel.cpp:1:
the Accel.cpp :1
is "#include "

发自我的 iPhone

jasonlo0509, 2018-02-09:
@qingzengsong  @MinLiAmoy 
May I know what specifically should I change to solve this problem, thanks!
Jason Lo

rzhao01, 2018-02-09:
Looking online, the solution is to #include <cstddef> before any other headers.

closed 2018-02-11 by rzhao01

hex0102, 2018-02-14:
I added #include <cstddef> in every .cpp file in both **utils** and **accel**. It seems work.