cornell-zhang/bnn-fpga #6: /usr/include/features.h:367:12: fatal error: 'sys/cdefs.h' file not found #  include <sys/cdefs.h>

qingzengsong, 2017-08-17:
when I make sdsoc_build to generate FPGA bitstream.
One error has happened:

INFO: [SDSoC 0-0] Performing accelerator source linting for top
INFO: [SDSoC 0-0] Moving function top to Programmable Logic
ERROR: [HLS 200-70] Compilation errors found:
Pragma processor failed: In file included from /home/songqingzeng/bnn-fpga-master/cpp/accel/Accel.cpp:1:
In file included from /opt/Xilinx/SDx/2016.4/Vivado_HLS/lnx64/tools/gcc/lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/cstddef:42:
In file included from /opt/Xilinx/SDx/2016.4/Vivado_HLS/lnx64/tools/gcc/lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/x86_64-unknown-linux-gnu/32/bits/c++config.h:393:
In file included from /opt/Xilinx/SDx/2016.4/Vivado_HLS/lnx64/tools/gcc/lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/x86_64-unknown-linux-gnu/32/bits/os_defines.h:40:
/usr/include/features.h:367:12: fatal error: 'sys/cdefs.h' file not found
#  include <sys/cdefs.h>
           ^
ERROR: [SDSoC 0-0] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2016.4/Vivado_HLS/bin/vivado_hls /home/songqingzeng/bnn-fpga-master/cpp/accel/sdsoc_build/_sds/vhls/top_run.tcl'
sds++ log file saved as /home/songqingzeng/bnn-fpga-master/cpp/accel/sdsoc_build/_sds/reports/sds_Accel.log
ERROR: [SDSoC 0-0] Build failed

Makefile:21: recipe for target 'Accel.o' failed
make: *** [Accel.o] Error 1

I don`t known the reason of this.

I think the file (/usr/include/features.h) shound not been bulid by sds++.
May be the evn PATH is error.
Thank all.

qingzengsong, 2017-08-17:
Pragma processor failed: In file included from /home/songqingzeng/bnn-fpga-master/cpp/accel/Accel.cpp:1:
the Accel.cpp :1
is "#include <cstddef>"
 .

urock, 2017-12-26:
For me the following worked

`sudo apt-get install gcc-multilib g++-multilib
`
https://forums.xilinx.com/t5/SDSoC-Environment-and-reVISION/SDx-2016-4-HLS-compilation-errors-from-stdio-h/td-p/766447

rzhao01, 2018-03-07:
Closing this as it looks like a g++ library issue and there hasn't been further activity.

closed 2018-03-07 by rzhao01