cornell-zhang/bnn-fpga #7: run make -j4 for sdsoc_build makefile, xdlmcheck is not found

liyoungforever, 2017-10-05:
Hi！ When I run make -j4 command for the sdsoc_build makefile,  I have a issue.  I used Ubuntu 16.04 OS, 2017.1 SDSoc software. 

[image]

when I run xdlmcheck ap_sdsoc command, it shows the same problem with my SDSOC
[image]

Thank you!

rzhao01, 2017-10-05:
I don't think I've ever encountered this issue, but it looks like an issue with the SDSoC installation to me. It looks like the SDSoC tool is looking for a program and was not able to find it. Did you make sure to set up the environment for SDSoC?

LiYang66, 2017-10-25:
Thank you! My issue has been solved! Appreciated!

liyoungforever, 2017-10-25:
Thank you! My issue has been solved!

closed 2017-10-25 by liyoungforever