cornell-zhang/bnn-fpga #8: child process exited abnormally

Eason19960910, 2017-10-15:
Hi , my SDSOC version is 2017.2 on ubuntu 16.04 with g++4.8

While executing "make -j4" in "cpp/accel/sdsoc_build"
I've met the following error

`sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../accel_test_bnn.cpp -o accel_test_bnn.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../Accel.cpp -o Accel.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../AccelSchedule.cpp -o AccelSchedule.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../AccelTest.cpp -o AccelTest.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
Processing -sds-hw block for top
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
Using system configuration -sds-sys-config linux
Using system configuration -sds-sys-config linux
Using system configuration -sds-sys-config linux
Using system configuration -sds-sys-config linux
child process exited abnormally
    while executing
"exec which [apc_get_global APCC_TOOLCHAIN]g++"
    (procedure "sdscc_set_toolchain_config" line 27)
    invoked from within
"sdscc_set_toolchain_config $toolchain_config $toolchain_type $proc_type $toolchain_name"
    (file "/home/wyx/Xilinx/SDx/2017.2/bin/xdcc.tcl" line 3581)
`
Is it something wrong with the SDSOC?
Thanks!

closed 2017-10-15 by Eason19960910

manymuch, 2018-01-12:
I have encountered the same problem,my SDSOC version is 2016.4 on ubuntu 16.04.3 with g++4.8.
How to solve this? Thanks!

rzhao01, 2018-01-14:
I've never encountered this error, sorry. The key part seems to be "Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux". Are you sure you are using a distro of Linux which is supported by SDSoC?

manymuch, 2018-01-15:
Thanks! I have sovled this problem. I think it's related to lack of  SDSoC enviroment variables.