cornell-zhang/bnn-fpga #9: Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2017.2/Vivado_HLS/bin/vivado_hls

Lishiqing, 2017-10-20:
I have this problem when I run the `make -j4` in the sd_soc_build folder.
The concrete error is like:
[image]
Is there someone who encountered this?
My system is Ubuntu 14.04, and the SDSoC is 2017.2

rzhao01, 2017-10-25:
We're trying to get SDSoC 2017.2 installed on our servers so we can look into this issue. Do you think this could be related to the tool? E.g. can you build the examples from Xilinx without any issue?

Lishiqing, 2017-10-26:
@rzhao01 Thank you for your responce! I think this is not related to the tool, but is related to the environment variable.Because I have encountered another problem.It's when I don't source the `setting64.sh`, I can compile the cpp step successfully.While I source `setting64.sh`, there will be an error `cc1plus: error: unrecognized command line option ‘-std=gnu++11’`.So am I wrong in the cpp compling step?

rzhao01, 2017-10-26:
The cc1plus error is your C++ compiler saying that it does not support the C++11 standard. You need a certain version of g++ (4.7 or later). I'm not certain, but maybe sourcing setting64.sh changes the g++ in your environment. Trying manually changing your g++ after you source it.

Lishiqing, 2017-10-26:
@rzhao01 OK! I will check it out

Lishiqing, 2017-10-30:
@rzhao01 I have changed my SDx to 2016.4 and run the command.Everything is OK.But I get this error:
[image].Is this a error about code?

rzhao01, 2017-10-30:
Is there more info on the error in sds_AccelTest.log?

Lishiqing, 2017-10-31:
@rzhao01 It seems to be a error about my clang compiler.Sorry for bothering you and thanks for your help!

closed 2017-11-02 by Lishiqing