cornell-zhang/bnn-fpga #11: how can i get the files in the params

Sun-xiaohui, 2018-01-19:


rzhao01, 2018-01-19:
In addition to the weights, each layer has 2 parameters associated with batch normalization. These are the k and h constants in the paper. So each layer requires 3 arrays, 3x9=27 arrays. I'm not sure what you mean by how to "get" the 27 files. If you run the get_params.sh script it should download the parameters archive.

Sun-xiaohui, 2018-01-20:
thanks  for your reply
what i mean how to “get”  is that if i want to extract these 27 files from https://github.com/MatthieuCourbariaux/BinaryNet  source code  , instead of getting the extracted files by running the corresponding script . Due to the improvement of the model, different parameters have been obtained. Now we want to redraw the corresponding parameters. How should we operate?
Thanks.

rzhao01, 2018-01-21:
You'll need to train the parameters first. In Courbariaux's code each layer has weights, biases, and 4 batch norm parameters. The biases can be ignored, and you need to calculate k and h from the batch norm parameters. Then package the arrays into a zip archive.

EmiyaLJ, 2018-01-22:
@rzhao01 
Hi, I'm currently working on a BNN model based on Pytorch. I'm sure the model is well trained now.
If I want to use your code to test the model's accuracy on FPGA, what should I do first ??  I am wandering what do the files in data folder mean? 
I have stuck with the problem for while, plz help me.

Thank you very much.

rzhao01, 2018-01-22:
The files in the data folder are the images from the CIFAR-10 dataset, as well as intermediate feature maps for the BNN we trained. If you are testing on CIFAR-10 you shouldn't need to change the data. You'll have to find a way to create the parameters archive files.

I think you'll have difficulty using our code unless your network is similar to the Courbariaux BNN.

Sun-xiaohui, 2018-01-24:
@rzhao01 thanks for your reply
thanks for your help, i have use the numpy to save the parameters in the form of the npy files. But how can i transform the file.npy  to the binary file . I use the numpy  save and savez methods to save the result, all with the npy suffix.  The result i use numpy to save is the same as the data in the npz archive. But what puzzled me was the fact that zip archive files were converted from npz files, or was there any other way to extract binary files without a suffix. Please answer my question. thanks.

ArchieGu, 2018-01-24:
@rzhao01 
Hi, I'm using the VGG net to train Cifar-10 dataset, which is quite similar to the Courbariaux BNN.
But I have modified some of the layers and add dropouts layers in the net, so I think I might need to change the files in data folder. Am I right?

rzhao01, 2018-01-24:
@Sun-xiaohui I added a script to convert npz files to zip files.

@ArchieGu No, the data folder is for input data (such as images to the CNN, or intermediate feature maps for per-layer testing). The params folder contains the weights and other parameters for the CNN.

Sun-xiaohui, 2018-01-25:
@rzhao01 
thanks for your reply
I use the numpy‘s  tofile method to convert npy files in the npz package you provided in the parameters folder. After conversion, each file is the same size and without the suffix. Then i use the ZIP command to package the file, but the size of the file after the package is smaller than the zip file you provided a few dozen KB, i will replace the zip file you provide the original zip file and copy the file onto an SD card, loaded into the board running time reported an error "accel_test_bnn.exe: /home/song/My_BNN/cpp/utils/ZipIO.cpp:36: unsigned int get_current_file_size(unzFile): Assertion `fsize > 0' failed.
Aborted".
 I do not know what is wrong with this process. Please give me some instructions. Thank you.

Sun-xiaohui, 2018-01-25:
@rzhao01 
and also i use the zipfile method to package the file, and the file size is the same as the original zip file, but when i copy onto an sd card,  when running  i get this error "ERROR: application performed illegal memory access and is being terminated". I think the reason why the previously compressed file is smaller may be caused by not use the zlib compression method. This time i use the zipfile method to compress the file to get the same size, but the operation is still an error, please help me to explain why. thanks.

ArchieGu, 2018-01-26:
@rzhao01 
Here is my question: The VGG net and resnet I am using have some slight differences from Courbariaux BNN (But the whole framework is similar) For example, in VGG net I use BinarizedLinear layer instead of the FullConnected Layer.  So in this case do I need to change the input in the data folder？？

Thank you for your reply.

rzhao01, 2018-01-26:
@Sun-xiaohui I'm not sure what you mean by zipfile method. You should dump a npy archive and use the script in my repo to convert it to a zip archive.

@ArchieGu The data folder is for input data to the network. Unless you changed the dataset you don't need to change the input in the data folder.

ArchieGu, 2018-01-26:
@rzhao01 
Oh, I see. Thank you so much.

ArchieGu, 2018-01-29:
@rzhao01 
What if I want to change the dataset to cifar-100, how can I modify the files in data folder???

rzhao01, 2018-01-29:
@ArchieGu I think CIFAR-100 would work, since it's the same dimensions as CIFAR-10. You'll have to replace the files _cifar10_test_inputs.zip_ and _cifar10_test_labels.zip_.

It will be fairly difficult to change the model to a ResNet, I'm not sure if the accelerator can handle the residual connections.

ArchieGu, 2018-01-30:
@rzhao01  
You've been great help, thank you so much.
If I need to change the model to ResNet, I need to modify the code related to the accelerator, right?
The _cifar10_test_input.zip_ and _cifar10_test_labels.zip_ are changed from .npz files to .zip files same as the files in param folder??

My question is how can we get these two files??

rzhao01, 2018-01-31:
Yes, to support ResNet you'll have to modify the accelerator.

For the data zip archives: the inputs are just the images of the dataset stored as a single float array (1 pixel per float), then zipped; the labels are the same with 1 label per float. I'm actually not sure why it was formatted like this. The code which reads the inputs/labels are in cpp/accel/DataIO.cpp and you can write your own version if you want. This is strictly host code and the accelerator doesn't interact with the zip archives.

Sun-xiaohui, 2018-03-06:
@rzhao01  
thanks for your help.
      I tried a new cifar100 dataset on the board, and I first replaced the test dataset section in the data folder and then looked at the results of the run,  identifying a high error rate. Then I extracted a new training parameter from the training network, replaced the zip file in the params folder,  and the results showed that the error rate was reduced, but the accuracy of the recognition was still different from the training.
      Now is the data folder cpp_ * These nine documents on the final operating results have an impact? These nine documents were expressed conv and dense layer of output data, or input data?  Now I want to ask you how i can replace these files, and I hope you can give some hints.
thank you.

rzhao01, 2018-03-06:
The only files in the data folder which matter to the classification is cifar10_test_inputs.zip and cifar10_test_lables.zip. The other files are **intermediate** layer data which is used for the per-layer testbenches.

Due to quantization and etc, there will be some accuracy differences between software (Python) and FPGA. But without details I can't say much more.

Sun-xiaohui, 2018-03-07:
@rzhao01 
ok, thank you so much

closed 2018-03-09 by Sun-xiaohui

wshbt2012, 2018-06-16:
Hello,
As you said before, in Courbariaux's code each layer has weights, biases, and 4 batch norm parameters.  So could you please give some hints about how can I calculate k and h from the 4 batch norm parameters?
Really appreciate.

rzhao01, 2018-06-16:
The calculation is Equation 4 in our [paper](http://dx.doi.org/10.1145/3020078.3021741).