cornell-zhang/bnn-fpga #13: Timing constraints are not met

hex0102, 2018-02-17:
Hi all, after I build the FPGA bitstream with the default project, I always got Timing constraints are not met. Is there any possible solution to this issue?

rzhao01, 2018-02-17:
The design does not meet timing for us either, however it still works on the board.

hex0102, 2018-02-18:
@rzhao01 any clues to this issue? And what's the working frequency of the accelerator?

hex0102, 2018-02-19:
@rzhao01 Thanks, I can also deploy the accelerator to ZC702 board now.  I'm still curious why the sdsoc reports "timing not met".

rzhao01, 2018-02-19:
SDSoC and other HLS tools use a timing model which is pessimistic. Failing timing in the tool does not necessarily mean failing on an actual board (it may fail on some boards and work on others). If timing is met in the tool then the design definitely works on the board.

closed 2018-04-05 by hex0102