cornell-zhang/bnn-fpga #14: question about the PIX_PER_PHASE parameter

hex0102, 2018-02-23:
PIX_PER_PHASE seems to be important in the accelerator.
And it's 2*32*32 in accel.h.
Could you clarify how the value is determined?

rzhao01, 2018-02-28:
The accelerator works in phases, each phase is a single iteration of  the loop in lines https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/Accel.cpp#L251-L252

In each phase, each convolver will process PIX_PER_PHASE pixels. This means they read in this number of input pixels and produce this number of (intermediate) output pixels. Each output pixel must be buffered. Increasing PIX_PER_PHASE should slightly improve performance but also slightly increase resource usage.

The structure of the BNN puts some constraints on what PIX_PER_PHASE can be, I don't remember too much here but you should run all the tests if you change PIX_PER_PHASE. I know you can reduce it to 32*32.

hex0102, 2018-03-15:
In the BIN_CONV function, two input images (convolver=2) are processed in parallel and every two input images are processed sequentially, so it seems that even though PIX_PER_PHASE increase the fix_buffer size, the inputs are still processed sequentially. Is that right?

rzhao01, 2018-03-16:
Yes, that's right. More images are processed in parallel (up to 8) but there will still be sequential processing.

hex0102, 2018-03-19:
@rzhao01  Hi, I find some comments in the codes confusing. What does the loop epilogue do in line 265? and why ignore conv results after processing the first word in line 348?

hex0102, 2018-03-20:
Are "loop epilogue"  and "ignore first conv results" related to the issue of padding?

hex0102, 2018-03-20:
I looked into the codes more. Are "loop epilogue" and "ignore first conv results" related to the issue of padding?

rzhao01, 2018-03-21:
Sorry for the late response. The [process word function](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/Accel.cpp#L93) contains three parts: Prologue (which updates the top row of the line buffer), Convolution, and Update (which updates the bottom 2 rows of the line buffer). Because of this, the first time we run Convolution, the Update hasn't happened yet, so we must ignore the results of the first Convolution. In general, we need n+1 calls to process_word in order to fully finish n words.

hex0102, 2018-03-26:
Thanks very much, it works.

hex0102, 2018-03-29:
@rzhao01  Hi, in the code, the number of Convolver is set to 2. Does this mean the two convolvers are working in a parallel way?  However, there are no explicit directives for parallel computing in opt.tcl. Is the parallel computing automatically enabled by the HLS tool?

rzhao01, 2018-03-29:
Yes, the convolvers are parallelized. There are loops which go up to the number of convolvers (key one being [here](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/Accel.cpp#L331)), and there are UNROLL directives in opt.tcl to unroll these loops.

The HLS tool does unroll loops based on a heuristic (static bounds, small bounds, small loop body, etc) but we should be unrolling all the key loops explicitly.

hex0102, 2018-04-03:
Thanks!  I have another question. Why inline conv3x3b in opt.tcl instead of pipelining it?

rzhao01, 2018-04-03:
conv3x3b is called by other functions, and those other functions are pipelined. So regardless of whether it is inlined conv3x3b will be pipelined.

If conv3x3b is not inlined, it will become a separate module and the II of calling function will be restricted by the latency of conv3x3b. By inlining the conv3x3b we ensure it does not constrain the II of the caller.

rzhao01, 2018-04-03:
Accidentally closed issue while typing.

We had to outline conv_word because inlining it caused the tool to take much longer to synthesize. I think conv_word is pipelined to II=1 it doesn't restrict the throughput of the main pipeline.

hex0102, 2018-04-15:
Hi, I've implemented multiple output channel (e.g. 4) inference in bin_conv in a pipelined manner, but the synthesis reports show excess TOP_MUX created which took  ~800% more LUTs than available. Do you have any clues about this problem?

rzhao01, 2018-04-16:
Multiple output channels is resource-inefficient because for each output channel we need a datapath which accumulates and binarizes the integer intermediate results. This datapath will need to be high-bitwidth (e.g. 12-bits). I would guess this is the issue, you should look at some of the HLS reports to see what consumes the most resource.

hex0102, 2018-04-18:
Does f_in in the paper refer to the number of convolver in the Accel.h file? 
I deployed the accelerator successfully, but the reported resource usage is different to the ones in the paper. Is this because of the different Vivado and SDSOC version (I use 16.04 version)?

rzhao01, 2018-04-18:
f_in refers to the number of convolvers, yes. Are you using the optimized branch, not the master (debug) branch? How much does the resource usage differ from the paper?

hex0102, 2018-04-19:
I'm using the master branch. And for the Convolver = 2 case,  the resource usage is far more than in the paper. 
+-----------------+---------+-------+--------+-------+
|       Name      | BRAM_18K| DSP48E|   FF   |  LUT  |
+-----------------+---------+-------+--------+-------+
+-----------------+---------+-------+--------+-------+
|Total            |      114|      4|   20329|  51779|
+-----------------+---------+-------+--------+-------+
|Available        |      280|    220|  106400|  53200|
+-----------------+---------+-------+--------+-------+
|Utilization (%)  |       40|      1|      19|     97|
+-----------------+---------+-------+--------+-------+

rzhao01, 2018-04-19:
That's because the master branch contains extra hardware for debug purposes. You should use the optimized branch.

hex0102, 2018-04-19:
@rzhao01  Thanks!  Is the correct data reported in _sds/reports/sds.rpt?   I used the data in sds_Accel.rpt, which tends to be large.  
P.S: the results from both optimized and master branch in sds.rpt could match the results in paper.
 (within 6%)

closed 2020-01-03 by zhangzhiru