cornell-zhang/bnn-fpga #15: what is contained in the cifar10_test_input.zip and cifar10_test_label.zip?

ArchieGu, 2018-03-12:
Hi, Ritchie Zhang @rzhao01 

I am working on the cifar-10 classification project based on Pytorch. And I got a question that do I have to change the data in these two files due to the shift of platform?

Second, I am wandering if  you can tell me  how do you create these two files? I want to  create my own test input files but I don't know how. 

Thank you so much.

rzhao01, 2018-03-12:
The data files simply contain the input and labels as floats in a single contiguous array. For the inputs this array contains all the test images, flattened. For the labels this array contains all the labels (non-one-hot) in order.

The easiest way to create these two zip files is to prepare the inputs/labels as numpy arrays, save them as .npz archives using `np.savez`, and convert the archives to .zip using the script [bnn-fpga/python/npz2zip.py](https://github.com/cornell-zhang/bnn-fpga/blob/master/python/npz2zip.py).

ArchieGu, 2018-03-13:
So, both Theano and Pytorch can use these two files if I just use the cifar-10 dataset, right?

We shift the theano version code into Pytorch version, and try to export all the parameters corresponding to your fpga code(we replace the weight,k,h in the param folder). However, the test accuracy is around 80% (test accuracy on PC is around 91%), which is lower than your original result. Could you please tell me what might be the cause of this lowered accuracy? Could this be a result of an incorrect procedure? If so, where might the problems be?

rzhao01, 2018-03-13:
Without more details I won't be able to help you. 20% accuracy sounds like a bug of some kind.

ArchieGu, 2018-03-13:
Hi, Ritchie @rzhao01  
https://github.com/ArchieGu/BinaryNet.pytorch
This repo is based on Itay's code, and we have done some slight changes on it, but the framework is remain unchanged.

In ./models/vgg_cifar10_binary.py we extract the parameters(k,h,weight) and convert them into .zip files. We replaced the files in the param folder and  we got 20% accuracy.  Could you please tell me what might be the cause of this? 

Looking forward to your suggestion.

ArchieGu, 2018-03-14:
Hi, Ritchie @rzhao01

I happened to come up with the same question as well.Here’re my thoughts after reading your post, though they might not be correct 

In order to create our own cifar-10 input test files, the first thing we need to do is to use Theano to read the original cifar-10 dataset.
Second, we seperate the lable from the image itself, and convert them into two arrays. One for images and another for lables. (I'm not quite sure how to do this, using PIL function or something else?)

Could you tell me if the two data files are extracted by the Theano version codes written by Courbarieaux ? Or you write a script to extract the data?

rzhao01, 2018-03-14:
I made a small change to Courbarieaux's code to generate the files. All I did was take the test images and test labels vectors, and use `np.savez` to save them as .npz archives.

Mukefan, 2018-03-15:
@ArchieGu 

This saves the data and tags used for the test. In Courbarieaux's code, it has been assigned to the parameters. The corresponding X,y attribute values are saved and are the desired data.

@rzhao01 
I now want to retrain the new data set based on the Courbarieaux's code, but the data set does not belong to the cifar series. The code corresponding to the zed board has been adjusted. But on how to save the corresponding parameters in the code of Courbarieaux's,  the current progress is slow, I hope you can give pointers. 
Thank you.

ArchieGu, 2018-03-15:
@Mukefan 
I am using the Pytorch version code. The model is very close to Courbarieaux's model. So I want to know if I need to rewirte the input_test files?

Testing the Pytorch version parameters while using the Theano version input test files cause low accuracy (20%)

rzhao01, 2018-03-15:
@Mukefan 
Courbarieaux's code uses 'lasagne.layers.get_all_param_values()` to retrieve the parameters and `np.savez` to save them. I also used this method. I know I can change the dataset (to MNIST) and use the same method to save parameters.

Mukefan, 2018-03-16:
@rzhao01 
thanks  for your help.

ArchieGu, 2018-03-20:
Thank you for your support！

closed 2018-03-20 by ArchieGu