cornell-zhang/bnn-fpga #16: Why the fpga-code only predicts one label???

ArchieGu, 2018-03-20:
[image]

As you can see, the code only predicts label 7, and we don't quite sure where the problem is.

We use the parameters extracted from Pytorch. And we also changed the input value range [-1,1] as the Theano version did.

rzhao01, 2018-03-20:
Is this the code I provided or did you make changes? Please be clear in the question as there are other people reading these threads.

In general this indicates improper importing of the weights or data.

ArchieGu, 2018-03-26:
@rzhao01 
Hi, sorry for bothered you again.
1. I didn't make any changes to your code.
2. Yes, I used my own parameters extracted from Pytorch.
3. The cifar10 images your provided is already in the range[-1,1], because it is the default processing range for Theano. In Pytorch, the default range is [0,1], so I have to make some changes to the images.

ArchieGu, 2018-03-27:
[image]

The method I use is provided by Pytorch tutorial.

rzhao01, 2018-03-27:
Oh you made changes in your PyTorch script. I thought you changed the input range for my provided code. You'll have to tell me exactly what you changed in my code for me to help.

I'm not familiar with PyTorch or your training code so I can't comment on that.

ArchieGu, 2018-03-28:
OK, you've been great help, thank you so much~

Sun-xiaohui, 2018-04-03:
@rzhao01  
Thanks for your answer.
In the code corresponding to the indicated line, [here](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/accel_test_bnn.cpp#L92),  there is only one  **binarize_input_images** function for the input data.  However, in the implementation of the function on [this line](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/AccelTest.cpp#L152),  there is no **binary method statement** or the **applying (x+1)/2 scaling to each element methods** as you are talking about.
How is this achieved?
thank you.

Sun-xiaohui, 2018-04-03:
@rzhao01  
Thanks for your answer. 

1. In the code that handles input data,  [here](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/accel_test_bnn.cpp#L92),  there is only one  **binarize_input_images** function for the input data.  However, in the implementation of the function on [this line](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/AccelTest.cpp#L152),  there is no **binary method statement.** How is it achieved?
2.  **whether** the **image and label**  in the **data file** and the **weight,k,h**  in the **params** must be **extracted from the same framework**. Because the distribution range of image is not the same among different frames. The default of the pytorch is [0,1], and Matth.‘s ’theano code  is [-1,1].
3. The distribution of the training data will affect the final identification of the board.

thank you.

ArchieGu, 2018-04-04:
@Sun-xiaohui 
I think what @rzhao01  means is that if your input is in range of [-1,1], then nothing need to be changed, if your input is in range of [0,1] then reshape your input into [-1,1] and continue with the original code or you can use 2*x -1 to modify the fpga code to read the input in range [0,1]. After this you can use your parameters to test the code.

Sun-xiaohui, 2018-04-04:
@ArchieGu 
Yes, whether you choose to add processing during training or processing on the code side of the board, the purpose of  adding the formula conversion is to make the data range when training and the data range when testing on the board be the same.

Sun-xiaohui, 2018-04-08:
@rzhao01 
I use the code provided by Theano version of Matth's, just add the corresponding 
```
  params = lasagne.layers.get_all_params(model)
        for param in params:
             if param.name == "W":
                 np.save(param.name,param.get_value())
```
 method in the main part of the training, other parts have not been changed, saved the parameters of 500 times trained out. 
However, after replacing the trained parameter file with the file in the param on the corresponding board, the error rate of the test result always fluctuates from 50% to 60%. I  tried many training results and could not match the low error rate during training. Is there any detail that needs attention?

rzhao01, 2018-04-08:
Did you make neccessary changes to the parameters? For the accelerator we make two changes: (1) remove biases, and (2) transform the batch norm parameters.

Sun-xiaohui, 2018-04-09:
@rzhao01 
Thanks for your reply.
Yes, I reconfirmed the program code again, where the bias added a new parameter **b=None**. 
The file in the corresponding save path contains a total of **54** files. Each of the **six** files corresponds to **w ,b ,beta ,gamma ,mean ,inv_std**.  The parameter file in the corresponding Batchnorm is processed as follows:
```
beta = np.load("./theano/arr_2.npy")
gamma = np.load("./theano/arr_3.npy")
mean = np.load("./theano/arr_4.npy")
inv_std = np.load("./theano/arr_5.npy")

k = gamma / inv_std
h = beta - mean * gamma / inv_std
```
Take all the files to get **w, k, h**, and save them as the corresponding 27 files in order. Is there anything wrong with my two-step operation?
I saved these changes again, and trained again. When the results come out, I share the latest results with you.
Thanks.

rzhao01, 2018-04-09:
**inv_std** is the reciprocal of the standard deviation, you should be multiplying instead of dividing by it.

You can test whether your kh calculation is working in Python. Simply override BatchNormLayer in Lasagne, define k and h parameters, and have the layer return `input * k + h`. Then you can test your modified parameters.npy before importing it in C++.

ValarSun, 2018-04-10:
@rzhao01 
Thanks for your reply.
Yes, it should  be multiplied by the **inv_std** variable. 
Why is the parameter k, h extracted from the corresponding batchnorm not directly used,  but also try a test?  The parameter variables obtained are consistent with the data dimensions in the params you provide. Is the process of data participation still different?

The code for batchnorm corresponding to k and h is as follows.
```
class MyBatchNorm(lasagne.layers.BatchNormLayer):
  def __init__(self, incoming, axes='auto',k=Constant(1), h=Constant(0), **kwargs):
    super(lasagne.layers.BatchNormLayer, self).__init__(incoming, **kwargs)
    if axes == 'auto':
      # default: normalize over all but the second axis
      axes = (0,) + tuple(range(2, len(self.input_shape)))
    elif isinstance(axes, int):
      axes = (axes,)
    self.axes=axes
    self.k = k
    self.h = h

  def get_output_for(self, input, deterministic=False, **kwargs):
    param_axes = iter(range(input.ndim - len(self.axes)))
    pattern = ['x' if input_axis in self.axes
               else next(param_axes)
               for input_axis in range(input.ndim)]
    k = self.k.dimshuffle(pattern)
    h = self.h.dimshuffle(pattern)
    #k = self.k
    #h = self.h
    # normalize
    normalized = (input) * k + h
    return normalized
```

The **input** here is a four-dimensional array **[num,Channels,row,columns]**, but the corresponding **k** and **h** are one-dimensional constants, and both the gamma and beta parameters are Consistent, but in lasagne's official source code, gamma, beta can be substituted into the calculation.  However k, h into the formula calculation will appear dimension mismatch problem, there is some confusion. What is the role of this batchnorm verification?
thanks.

rzhao01, 2018-04-10:
I don't quite understand your question - you'll have to explain what "data participation" means and what dimensions are mismatched.

Batch norm parameters are supposed to be 1-dimensional, each output feature map has a k and h, so k and h should be arrays with length equal to the output channels.

I suggested verifying the batch norm because you mentioned testing on the board - I wanted to make sure you tested the parameters in the Python script before importing them to FPGA.

rzhao01, 2018-04-11:
No problem. So you made sure k and h worked in Python? Are you still having an issue on the FPGA board?

ValarSun, 2018-04-11:
Yes, the converted data works on both Python and FPGA board.
For the C++ code part you provided, after make, there are two versions of the executable program, which are cpu and fpga. But if you want to generate a program that can be run directly on the PS side of the FPGA chip, directly on the corresponding arm, without hardware acceleration, what efforts should I make?
Thanks.

rzhao01, 2018-04-11:
Good to hear!

You just need a software (C++) implementation of the BNN, so instead of marking top with the HLS pragmas in Accel.h, just write normal C++. In fact, the `master` branch has some envvar switches to run the dense layers in software: [here](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/accel_test_bnn.cpp#L23-L24)

ValarSun, 2018-04-11:
OK, you've been great help, thank you so much~

rzhao01, 2018-06-16:
Closing due to lack of activity. All questions seem to have been answered.

closed 2018-06-16 by rzhao01