cornell-zhang/bnn-fpga #18: Error about the SDSoC license

wangj346, 2018-05-04:
Hello, when I make the Makefile in the ' cpp/accel/sdsoc_build ' , errors about the SDSoC license occurs.
[image]
I have get the corrected license and it show 'yes' in the Xilinx license manage. 
(by the ways, I am using the SDx and set the $PATH to the SDX/bin)
Any advices will be appreciated.

wangj346, 2018-05-04:
Hello,I have fix the problem by rewriting the ~/.bashrc and rebooting.But I come across another problem.
[image]
May I get some suggestions?

rzhao01, 2018-05-04:
I'm not sure what the problem is. You might have to look at the log file (Accel.log if it exists) so see if there are other messages.

rzhao01, 2018-11-26:
Closing due to lack of activity.

closed 2018-11-26 by rzhao01

Harley-Quin, 2023-05-10:
Hi,
how did you resolve the first question? I met the same question. Thank  you.