cornell-zhang/bnn-fpga #19: Can I change the code for CNN?

ArchieGu, 2018-05-06:
@rzhao01 
Hi, sorry to bother you again.

Any chance that the code can be shifted to CNN version??
Like I put some trained CNN weight and predict the accuracy? Running on ARM.

ArchieGu, 2018-05-06:
In your paper, you have compared the BNN model with CNN model, so I am wondering where can I find the program for CNN prediction? 
If you have the code, would you like to post it on Github??

rzhao01, 2018-05-07:
The hardware architecture for a BNN and CNN accelerator would be fairly different. It'll be difficult to just shift it to CNN.

The CNN accuracy is taken from other papers from ML researchers. You should look at those papers to see how they got their results. I can't replicated every paper I cite.

rzhao01, 2018-11-26:
Closing due to lack of activity.

closed 2018-11-26 by rzhao01