cornell-zhang/bnn-fpga #20: Timing problem

ShawnLess, 2018-05-29:
Hey all:
I installed the SDx 2017.2 tool in Ubuntu and was able to generate CSIM and almost finished the FPGA build.

However, I got this error when run 'make' under sdsoc_build and I didn't see the bitstream and uImage were generated.

Appreciate any suggestions. 

```bash
Finished 1st of 5 tasks (FPGA synthesis). Elapsed time: 00h 31m 01s.
.....
Finished 2nd of 5 tasks (FPGA logic optimization). Elapsed time: 00h 00m 40s.
.....................
Finished 3rd of 5 tasks (FPGA logic placement). Elapsed time: 00h 03m 18s.
.........................
Finished 4th of 5 tasks (FPGA routing). Elapsed time: 00h 04m 35s.

===>The following messages were generated while  Compiling (bitstream) opencl binary: bin Log file: /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/p0/_vpl/ipi/ipiimpl/ipiimpl.runs/impl_1/runme.log  :
ERROR: [VPL-1] Design failed to meet timing
WARNING: [VPL 60-732] Link warning: There is no resource utilization data in DSA, utilization DRC is skipped
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, route_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/opt/Xilinx/SDx/2017.2/bin/vpl  --iprepo /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/iprepo/repo  --iprepo /opt/Xilinx/SDx/2017.2/data/ip/xilinx  --platform /opt/Xilinx/SDx/2017.2/platforms/zed/zed.xpfm  --temp_dir /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/p0  --output_dir /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/p0/vpl  --input_file /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels top --webtalk_flag SDSoC'
sds++ log file saved as /home/ubuntu/workspace/bnn_fpga/cpp/accel/sdsoc_build/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:35: recipe for target 'accel_test_bnn.exe' failed
make: *** [accel_test_bnn.exe] Error 1
```

rzhao01, 2018-05-30:
I've only tried the design with SDSoC, not SDx. I know these are different tools though I'm not sure what the differences are. Can you try compiling with SDSoC?

ShawnLess, 2018-05-30:
Thanks you again,  Ritchie.  We definitely should try SDDoC.

rzhao01, 2018-11-26:
Closing due to lack of activity.

closed 2018-11-26 by rzhao01