cornell-zhang/bnn-fpga #23: ported to ZYNQ UltraScale -- only predicts one label

conjam, 2018-11-26:
Howdy y'all,

I have been trying to port this over to the UltraScale. It seems to be going alright, but when I run the application on the board, I predict mostly class 1 and class 2. I have made no changes to the code, other than compiling libraries in aarch64 to be compatible with the UltraScale.

I imagine there is some typing I'm overlooking, but I can't seem to pick it out. Has anyone encountered this?

rzhao01, 2018-11-26:
I've never worked with UltraScale, so I don't know the problem off the top of my head. I would say look through the logs to see if there's any strange warnings or messages, and try with CONVOLVERS=1 to see if that works.

empissas, 2018-11-30:
I have the same issue . I try to port the BNN  in the arm(64) processor of  a zcu102 board  by cross compiling with aarch64-linux-gnu-g++ compiler  and  it can't predict any label. The error seems to be in the fp_conv function where the output error rate is very high. Could you please suggest any solution ?

rzhao01, 2018-12-12:
Sorry for the late reply. I would try running with native data types instead of Vivado HLS types to see if that's a problem. [Accel.h](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/Accel.h) contains the typedefs for the loop bounds and [Typedefs.h](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/utils/Typedefs.h) contains the typedefs for the data.

empissas, 2018-12-12:
As i noticed ,   the arm processor  sometimes cannot  access " the 32  high  bits"  (63 down 31)  to  assign a    value   .  Perhaps a code with native data types  may work without this problem , but how  ( for example an ap_fixed type )  would replaced ?

empissas, 2018-12-12:
Yes the problem is only in the arm53 processor when assigning values to  the kh mem and dmem. Weights seems to be fine because the values are written ( bit by bit ) . Moreover  ( the bin conv  and  the dence  functions  when execute in the arm  doesn't produce the appropriate results )

rzhao01, 2018-12-14:
I wonder if the problem can be solved by doing shift + mask instead of ap_int assignment. In the function which loads the input data I use ap_int assignment: [here](https://github.com/cornell-zhang/bnn-fpga/blob/master/cpp/accel/AccelTest.cpp#L164). Maybe try replacing that line with bitwise operations?

empissas, 2019-01-02:
Thanks for the reply . I will try to replace all these lines that cause that problem . 
Furthermore, as i noticed  bin-conv functions execute N times depending on the output fmaps (64, 128 etc). Could  these iterations  execute simultaneously or they have any dependence ?

rzhao01, 2019-01-02:
Each invocation of bin_conv is independent. From what I remember, we chose to parallelize across different inputs (CONVOLVERS) rather than outputs (bin_conv) to avoid duplicating the circuitry for batch_norm and so on.

empissas, 2019-01-22:
Thanks. Is the I/O time included in the timer for each accelerator  invocation , or the displayed time refers only to the execution time .

rzhao01, 2019-01-22:
The displayed time is wallclock time, so it includes the IO/.

empissas, 2019-02-18:
Does this time include the I/O from the external memory or it only includes the I/O between the accelerator and the on chip memory ?

Thanks

rzhao01, 2019-02-18:
It includes the I/O from external memory.

empissas, 2019-02-19:
Which is the frequency of the system ?

Thanks

rzhao01, 2019-02-19:
The clock period is 7ns so the frequency is 143MHz.

closed 2020-01-03 by zhangzhiru