cornell-zhang/bnn-fpga #24: Binarized yolo

Thilanka97, 2019-01-08:
@rzhao01 I am trying to run yolo on fpgas in a efficient way. I was thinking of maybe I can run binarized yolo on fpga using your work. will I be able to do this ? What changes do I need to make ? If I find a binarized weightsof yolo, then would it be possible ? please guide me. 

Thanks in advanced !

rzhao01, 2019-01-10:
Running YOLO with this design would be challenging. My design is targeted for CIFAR-10 and uses a specialized first layer with 32x32 inputs. YOLO uses much larger inputs (448x448). YOLO also uses a combination of 7x7, 3x3, and 1x1 filters while the BNN in my design only uses 3x3 filters.

I would say you'd be better off looking for a different BNN accelerator which targets ImageNet and modify that.

Thilanka97, 2019-01-11:
@rzhao01 thanks alot for your reply. helps alot!

closed 2020-01-03 by zhangzhiru