cornell-zhang/bnn-fpga #26: SDSoc Compilation Errors

HonggangYu, 2019-11-22:
Hi Ritchie,

I find some errors while compiling the Makefile in "bnn/cpp/accel/sdsoc_build" on the Ubuntu 16.04 system. Here is the log information:

sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../accel_test_bnn.cpp -o accel_test_bnn.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../Accel.cpp -o Accel.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../AccelSchedule.cpp -o AccelSchedule.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
sds++ -sds-pf zed -dmclkid 1 -sds-hw top Accel.cpp -clkid 1 -hls-tcl sds.tcl -sds-end -c ../AccelTest.cpp -o AccelTest.o -DHLS_COMPILE -I../../utils -O3 -std=gnu++11 -g
INFO: [SDSoC 0-0] Processing -sds-hw block for top
INFO: [SDSoC 0-0] Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
INFO: [SDSoC 0-0] Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
INFO: [SDSoC 0-0] Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
INFO: [SDSoC 0-0] Platform system configuration option -sds-sys-config was not specified, searching for a configuration that uses the specified OS linux
INFO: [SDSoC 0-0] Using system configuration -sds-sys-config linux
INFO: [SDSoC 0-0] Using system configuration -sds-sys-config linux
INFO: [SDSoC 0-0] Using system configuration -sds-sys-config linux
INFO: [SDSoC 0-0] Using system configuration -sds-sys-config linux
child process exited abnormally
    while executing
"exec which [apc_get_global APCC_TOOLCHAIN]-g++"
    (procedure "sdscc_set_toolchain_config" line 27)
    invoked from within
"sdscc_set_toolchain_config $toolchain_config $toolchain_type $proc_type $toolchain_name"
    (file "/home/yu/Documents/SDSoc/SDx/2016.4/bin/xdcc.tcl" line 3616)
child process exited abnormally
    while executing
"exec which [apc_get_global APCC_TOOLCHAIN]-g++"
    (procedure "sdscc_set_toolchain_config" line 27)
    invoked from within
"sdscc_set_toolchain_config $toolchain_config $toolchain_type $proc_type $toolchain_name"
    (file "/home/yu/Documents/SDSoc/SDx/2016.4/bin/xdcc.tcl" line 3616)
Makefile:21: recipe for target 'Accel.o' failed
make: *** [Accel.o] Error 1
make: *** Waiting for unfinished jobs....
Makefile:21: recipe for target 'AccelTest.o' failed
make: *** [AccelTest.o] Error 1
child process exited abnormally
    while executing
"exec which [apc_get_global APCC_TOOLCHAIN]-g++"
    (procedure "sdscc_set_toolchain_config" line 27)
    invoked from within
"sdscc_set_toolchain_config $toolchain_config $toolchain_type $proc_type $toolchain_name"
    (file "/home/yu/Documents/SDSoc/SDx/2016.4/bin/xdcc.tcl" line 3616)
Makefile:21: recipe for target 'AccelSchedule.o' failed
make: *** [AccelSchedule.o] Error 1
child process exited abnormally
    while executing
"exec which [apc_get_global APCC_TOOLCHAIN]-g++"
    (procedure "sdscc_set_toolchain_config" line 27)
    invoked from within
"sdscc_set_toolchain_config $toolchain_config $toolchain_type $proc_type $toolchain_name"
    (file "/home/yu/Documents/SDSoc/SDx/2016.4/bin/xdcc.tcl" line 3616)
Makefile:27: recipe for target 'accel_test_bnn.o' failed
make: *** [accel_test_bnn.o] Error 1

Could you please help me fix these compiler errors?

Thanks a lot for all your help.

rzhao01, 2019-11-24:
I'm not sure what the error is just from the log, try digging in some of the SDSoC logs to see if you can find a different error message. Which version of SDSoC and g++ are you using?

HonggangYu, 2019-11-26:
Hi Dr. Ritchie,

Thanks for your quick reply. In our implementation, we use the **SDSoC 2016.04** and **g++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609** to compile the Makefile in "bnn/cpp/accel/sdsoc_build" on the Ubuntu 16.04. I hope you can help me with this matter.

Thanks so much.